TeamOpenFIRE/OpenFIRE-Firmware #27: [REQUEST] Add the option to change D-Pad as Keyboard keypad

gustavoalara, 2024-09-18:
When you use the OpenFire lightgun in mouse\keyboard mode the D-Pad is assigned to the gamepad\joystick part (like the analog stick). However, emulators like Teknoparrot when configuring a lightgun game in raw-input mode doesn't detect the D-Pad or analog Stick and you can't assign these buttons in the game input. 

So, when in mouse mode at least D-Pad must be assigned to the up, down, right and left keys (if the stick could be assigned too, will be wonderful)

SeongGino, 2024-09-18:
In the short term, you can always build the firmware from source, changing the contents of `LightgunButtons::ButtonDesc[]` in `SamcoEnhanced.ino` to your preferred controls; in this case, just the entries for `bGunUp`/`bGunDown`/`bGunLeft`/`bGunRight`. Refer to `COMPILING.md` for info on the manual build process.

In the long term, this is where button function remapping would come in handy so it can be configured from the App.

Analog-to-digital will be handled in its own PR, as it's polled separately from the rest of the digital inputs so it would need its own decision tree.

gustavoalara, 2024-09-19:
Thank you @SeongGino 

First approach is the way I did, but as you said the button function remapping would be a nice feature to customize the keys people use in their games/emulators without the need of rebuild the firmware

PR #100 (2025-05-09): Add full remappable button mappings, Analog-to-Digital presets
Relies on newest App version (TeamOpenFIRE/OpenFIRE-App#72) to expose these settings.
Initial ButtonsDesc defined in `OpenFIREprefs.h` is now copied to the backbuffer introduced in #97 (and subsequently saved), and new methods directly read and write to this backbuffer. When reloading inputs map, this backbuffer is used for updating the main ButtonsDesc array. This (finally) closes #27 (without manually editing the descriptor in-code).

The `playerStartBtn`/`playerSelectBtn` blobs are now more context-sensitive - by using the fille [... cut]
closed 2025-05-09 by SeongGino