TeamOpenFIRE/OpenFIRE-Firmware #28: [TESTING NEEDED] Official/automated MiSTer support

SeongGino, 2024-09-26:
Hey all!

A new branch has been added with preliminary support for the *MiSTer FPGA* system.

For now, you'll have to refer to [these automated builds](https://github.com/TeamOpenFIRE/OpenFIRE-Firmware/actions/runs/11044872603) for the firmware to support this, as well as a [patched MiSTer executable](https://github.com/SeongGino/Main_MiSTer/releases/tag/OF-test) with the needed bits to both recognize OpenFIRE lightguns as well as automatically send the right command so they will work properly.

An overview of how this works and setup steps:
 - Download the patched `MiSTer`, and copy it to the root of your MiSTer bootable medium (either SD Card or USB Storage); you should be prompted to replace the executable.
 - Update your guns as usual, using the linked firmware built from the `MiSTer` branch - either by holding BOOTSEL when plugging in, or by clicking *Reboot to Bootloader* under the *Gun Tests* tab inside the OpenFIRE App. Drag'n'drop the `OpenFIREfw.uf2` for your board.
 - Boot up your MiSTer with your patched boot medium inserted, and connect an OpenFIRE gun. If your gun has a working OLED connected, then you will immediately notice the gun booting with the usual Mouse glyph now depicting a Gamepad icon. In any case, when the gun is connected, you should be able to access the global *Lightgun Calibration* menu in MiSTer by pressing F10 on a keyboard (this can be ignored).
 - Setup your lightgun's buttons - you only need to press any button for the first function, then undefine/skip all the rest, so that MiSTer saves a controller configuration for your lightgun.
 - On supported cores, set your controller port of choice to a lightgun controller - in PS1 for instance, this would be either the *GunCon* or *Justifier* controllers.
 - Setup your lightgun's core mappings, namely the trigger and whatever other auxiliary buttons are supported. You'll likely have to map the "Right" function first, but you can skip all the other buttons until you get to functions relevant to the emulated lightgun controller.
 - Finally, after the core has booted and you have a screen of full brightness (e.g. the PS1 BIOS), press F10 to go to the core's *Lightgun Calibration* menu. This is important, as **you MUST aim at the borders of the GAME IMAGE for 1:1 accuracy, NOT the edges of your TV.** As of now, MiSTer does not seem to scale inputs appropriately for the display, and this must be set anytime you change the scale of your displayed game image. *To reiterate, this is **not** an issue with OpenFIRE, but rather the inherent nature of MiSTer lightgun emulation.*

Some notes about the current implementation:
 * The way the new heuristics in the patch needed to detect the path of the gun's serial device is... a bit of a hack, to say the least. Currently, at a high level, the port detection logic looks for devices with the path `/dev/serial/by-id/usb-{underscoredName}_{trimmedID}-if00`, and then merges this with the actual echo command used and sends the resulting string to the shell. MiSTer performs all operations as root (yay!), but parts of the path are hardcoded (oof) and *may possibly* be prone to breaking under certain unknown conditions if this general path ever deviates. That being said, I have tested with multiple guns with no issues, but there is a chance it might break if the `if00` part of the path ever changes.
   * If you ever run into issues with guns not being auto-signalled, the version of the patched `MiSTer` executable will report any successful and failed device messages over the serial port. Refer to [this page on the MiSTer Wiki](https://github.com/MiSTer-devel/Wiki_MiSTer/wiki/Console-connection) for information about accessing UART, and make sure your current `MiSTer.ini` has `debug=1` set.
   * The current MiSTer_Main OF code can be seen @ SeongGino/Main_MiSTer@44a2e95 if you'd like to take a look, or are willing to offer less hacky suggestions before submitting a PR.
 * The MiSTer-side implementation has been tested on a RetroRemake (also more crudely known as the "Taki Udon") MiSTer Pi Mega Pack triple stack, which *should* be like-for-like compatible and behave the same as a DE-10 Nano + MiSTer Addons triple stack. Anyone who has a DE-10 Nano-based MiSTer triple stack setup, or a basic setup using other USB hubs are encouraged to test this as well to ensure it works across all setups.
   * Other more exotic setups, i.e. MiSTercade or consolized MiSTer boards have not been tested, and it would be nice to confirm that this hack works on those systems too.

As always, if you want to support further developments like this, [remember to support the team!](https://github.com/TeamOpenFIRE/.github/blob/main/profile/README.md) The support of individuals like you are how goals like this come to life.

Higgy69, 2024-11-15:
Hello.
I have original MiSTer and QMTech MiSTer. I am using the QMTech as Lightgun test bed.
I also use one of those Rii style mini keyboards, and it seemed with the OpenFIRE plugged in the keyboard would not work? I had to use hardwired USB keyboard. Removing OpenFIRE the Rii wireless keyboard started working again.
Going back to my findings, OpenFIRE is detected, I can define keys and F11 calibration (once F12 was pressed first, I was expecting to be able just to press F11 in Core).
But although Calibration seems to be ok, positioning does not work in Core. Button presses are ok.

I have a suggestion. Please remember you are the person with number 1 knowledge. 
Let us start off picking just one Core to test. I suggest NES and Duck Hunt because files are small.
Then you can show exactly the commands/options you are selecting. I find there are lots of options, Mouse, joystick etc etc. Remember you know exactly what you are doing, we are following so might miss something. Narrowing down the options will help the focus on results.
Many thanks.

Arakon, 2024-11-15:
I have successfully used the current firmware with the lastest specific OF-MiSTer build on a _MiSTer Pi_ setup in Time Crisis and Point Blank. Duck Hunt on NES also worked.

Saturn does not work, but that's probably cause the core currently only supports "real" light guns via SNAC. I tried using Mouse mode, but that didn't work either.

SeongGino, 2024-11-15:
@Higgy69 
> Please remember you are the person with number 1 knowledge. Let us start off picking just one Core to test. I suggest NES and Duck Hunt because files are small. Then you can show exactly the [...]

I *do not,* in fact, "know exactly what I'm doing" because I know as much as any beginner level MiSTer user, but on'y testing one core at a time defeats the point of the testing, as this isn't "testing for MiSTer NES core" - this is for *the entire MiSTer ecosystem.* And because all cores do their own things and have their own idiosyncrasies, I can't peer into the future and predict how every gun system core setup will go.

All I can provide are the generic setup instructions, which are accurate. It should be assumed that you've calibrated against the display you're using the MiSTer on from the gun's side; reading the whole post would've indicated as such near the bottom.

Is it confusing at times? Yes, but ***that's on the MiSTer interface as it exists, not us.*** I'm not here to fix MiSTer's UX; just making sure OpenFIRE works on it as smoothly as it can. Any other concerns can be voiced directly on `Main_MiSTer`, because I am not in control of that. 'tis are the plights of unpaid volunteer work.

Still, at least the heuristic seems to still work fine on other MiSTer boards. Is there anyone using a Terasic board, though, just to be sure?

@Arakon Thank you for your test results, happy to hear it works on your setup as well.

Higgy69, 2024-11-15:
Hello,

Sorry dyslexia/neuro diversity makes things difficult for me sometimes. I know with IR-GUN4ALL I asked you questions which met with a similar response to this one.
I would say the way our brains work, we are incompatible with each other 😄.
Never mind I still want to help.
I am sorry if what I wanted to say and what I wrote on my first comment were not the same thing. 
I usually print off instructions as I find paper easier to read, I did not have access to a printer  so sorry if I missed something. I just wanted to help as there were no replies from 26th September.

What I meant about you knowing what you are doing, is like when you give someone directions to a location. You may miss out something because you are familiar with the location, but to the person following the instructions they have no knowledge so they can only go by the given instructions.

You must realise I am not having a go at you, OpenFIRE team or MiSTer. I just want to help but I might need specific questions answered. It's how my brain works, otherwise I get stuck, yes the answer might be in the text but sometimes I can't see it. Sorry.

Putting my autistic hat on, on the second bullet point regarding OpenFIRE firmware update, this should also say Calibrate and Save within OpenFIRE UI if I understand what @Arakon has told me. Again just making sure I have understood correctly.
 This should be on the screen the MiSTer will also be used on, but my monitor is 16:9 so Windows calibration is in that aspect, but again for example, the NES Core is 4:3. 
The second to last bullet point about Calibration on the MiSTer not being feasible so then use a PC gives me mental conflict.

My point being for NES is that once I get one Core working, I will then not be disheartened when I can't get anything to work and I give up. Which is what happened with IR-GUN4ALL.
I've helped with loads of retro projects and get an enjoyment from helping the community move forward.

On the NES Core 'Periphery' has Zapper (Mouse)/Zapper (Joy1)/Zapper (Joy2)
'Zapper Trigger' has Mouse/Joystick
'Swap Joysticks ' has No/Yes

@Arakon would you please be as kind to say which settings you used. Then least I have the correct settings.

My original MiSTer has the Terrasic board so I can test that and hopefully be helpful ;) 

Sorry for the long post and additional explanations. But I really want to help and I know from past experience my brain can be incompatible with how other brains work, so it's best to explain.

Thanks

Higgy69, 2024-11-16:
@Arakon thank you for that information. I made sure calibrations were stored and a got NES - Duck Hunt working. 
Now I know I have a working system I will go through and test other Cores.

@SeongGino I can confirm that the Terrasic DE10 board works. 👍

I can go back and check everything on the QMTech board and look into that Rii wireless keyboard not working as it was ok on the Terrasic board. 

Thanks

Higgy69, 2024-11-17:
Hi,

I found some time for testing. This is all on a Terrasic MiSTer. I will try on the QMTech later.

Genesis Core - 'A' is trigger when remapping. Works with 'Menacer' games (Lethal Enforcers). I can't get Justifier games to work - Terminator 2 for example.

Megadrive Core - does not seem to work.

SMS Core - Works, but I can't seem to find how to set the Trigger. For example on 'Operation Wolf' I can only activate the Grenades.

SNES Core - Works with 'Superscope' games. Can't get 'Justifier' games to work. There is a Crosshair that moves with the gun but the controls are linked to the GCon2's D-pad (as per 'Genesis Core' for the Justifier). Games that use Justifier - Operation Thunderbolt and Revolution X.

Thanks

Higgy69, 2024-11-24:
PSX Core - working well. I played quite a few games. Some using GunCon and others using Justifier?

BrainGremlins, 2025-02-05:
I have a question about this. Some folks were talking about this in the mister discord the other day. Would it be possible to allow for a gamepad mode (possibly toggled via a button combo or the openfire app) and let the inbuilt functionality of the mister to handle mapping and such

SeongGino, 2025-02-05:
> I have a question about this. Some folks were talking about this in the mister discord the other day. Would it be possible to allow for a gamepad mode (possibly toggled via a button combo or the openfire app) and let the inbuilt functionality of the mister to handle mapping and such

Well for one, the whole point of this *is to automatically enable this mode* when plugged into MiSTer so that there's as little user intervention needed as possible. Because gamepad mode inherently is a workaround for a problem on PC, it shouldn't be a mode that can be accidentally enabled without the user explicitly engaging it, which is why it's activated via serial feedback only on PC (and having to toggle between input modes from a *configuration only desktop app* is a clunky and unacceptable UX).

For second, MiSTer uses a whitelist system, so this change has to be committed *anyways* to even allow OF guns to be seen as lightguns. It isn't as simple as "this gamepad says it's a lightgun, so use as lightgun" without MiSTer main being aware of it beforehand. And because this explicitly only works with and looks for guns using OF's specific vendor ID, it shouldn't conflict with any other input device that the MiSTer does support, so it shouldn't be a problem.

If you read the step-by-step instructions in the main issue text, you would also know that even *with* MiSTer support, it's nowhere near a seamless experience anyways, as it's how all USB guns have to be configured across MiSTer.

I wasn't made aware of any conversations being had regarding this (and haven't been mentioned either) so I don't know what "folks in the mister discord" are or what they could be saying.

BrainGremlins, 2025-02-05:
Sorry, didn't mean for this to undermine you or the work you're doing to make the best UX you can. I wasn't sure of the best way to draw attention to this sort of thing. I see you've talked to the developer who added lightgun support to saturn on the mister discord and cleared things up. Next time something like this occurs where it could be useful to talk to someone I could ping you on the discord instead?

SeongGino, 2025-02-05:
> Sorry, didn't mean for this to undermine you or the work you're doing to make the best UX you can. I wasn't sure of the best way to draw attention to this sort of thing. I see you've talked to the [...]

A ping would be appreciated if it's something regarding me or needs my attention, yes. :)

SeongGino, 2025-02-05:
Anyways, PR submitted to MiSTer Main. Will close this issue if/when it gets pulled in. :D

SeongGino, 2025-02-05:
https://github.com/MiSTer-devel/Main_MiSTer/pull/964 merged, closing.

closed 2025-02-05 by SeongGino