TeamOpenFIRE/OpenFIRE-Firmware #34: [REQUEST] Support 100k thermistors additionally to TMP36

Kalrkloss, 2024-11-11:
The TMP36 is a very precise, but also expensive part, and often hard to get.

It'd be much cheaper to support standard 100k thermistors. 
They're not only cheap, but also easy to get (also prewired), because they're used in many 3D printer hotends.

An example circuit and code can be found here: https://www.circuitbasics.com/arduino-thermistor-temperature-sensor-tutorial/

SeongGino, 2024-11-11:
TMP modules don't seem to be appreciably more expensive than like-quantity thermistors, so I don't really see the need to over-complicate things for the sake of <$1 savings. Not to mention I don't *have* any to test with.

Feel free to send a pull request, but it would need its own temperature reading function, and be added to the devices list and require a save table change.

Kalrkloss, 2024-11-12:
I'll test the temperature reading code on a bare RP2040, and then try to integrate it.