TeamOpenFIRE/OpenFIRE-Firmware #67: problems setting DUAL_CORE

alessandro-satanassi, 2025-04-03:
**Describe the bug**
From a 'virgin' micro, compiling with DUAL_CORE enabled, when I connect the micro to the OpenFIRE_APP App, the micro switches for a moment to the right communication mode with the APP (Docker, I think it's called - it displays the Open Fire icon on the display) and then immediately returns to the previous mode and no longer communicates with the App which remains open and the message "OpenFIRE Welcome - Pull trigger to start calibration" reappears on the micro. Before the latest numerous changes to the code, this did not happen. In SINGLE CORE, however, it works.
I tried to search for the problem, but I did not find the cause.
The problem occurs with both rp2040 and ESP32

**To Reproduce**
1 compile with DUAR_CORE active
2 upload firmware to the micro;
3 start App OpenFIRE App
4 select USB serial port
5 the micro switches to the right mode for a moment and then immediately exits from docker mode

**Expected behavior**
It should have worked, just like it works in SINGLE CORE (before the last several changes it worked fine)

**System Info**
 - OS: Windows 11
 - Microcontroller board: `rpipicow` and 'esp32s3'
 - OpenFIRE Firmware version: latest repository changes

**Additional context**

SeongGino, 2025-04-03:
The code builds with Dual Core *by default* and this hasn't been an issue in all of my testing with the current App code and on both Linux and Windows. What are you talking about?

closed 2025-04-03 by SeongGino

alessandro-satanassi, 2025-04-03:
I fear I have made a mistake, now it gives me the same problem both in SINGLE_CORE and in DUAL_CORE, so I fear that the problem is of another type, with all the latest changes and a thousand tests, I no longer know where the problem occurs, however now with the very latest commit of OpenFIRE Firware and OpenFIRE_APP, the problem always gives me. 
Well, I'll double check everything... maybe I messed up by mistake and didn't notice... if you like, the problem is me

SeongGino, 2025-04-03:
Tested with two separate boards, one even factory-resetted back to default values to recreate the same scenario, as well as had a third party test with the builds in a current PR. Seems like it's working fine here 🤷‍♀  
[image]

I already tested everything for the currently waiting PR last night and this repo's been in the same state as then. So it can't be an issue with our repo.

I don't want to sound harsh, but it's not our responsibility if a fork breaks when upstream is still working perfectly fine; that's largely in part your responsibility to resolve as a maintainer.

alessandro-satanassi, 2025-04-03:
If it works for you, the problem is certainly mine, but the changes for esp32 are so few and do not concern that area of ​​the code that I do not think they affect. Anyway, thanks for the advice and for the patience you have. Now I try to do everything on a linux machine and delete platformio cache, etc. etc.. .. I hope to solve it soon

SeongGino, 2025-04-03:
It doesn't matter how many changes there were for another platform.

You brought up ESP in the issue body. This repo specifically *is for the RP2040,* not the ESP. So when I see an issue in my RP2040 codebase, I expect *not* to see references to non-RP2040 platforms; if there is an issue, the onus is on you to *test with the codebase in **this** repository, with **this repo's actions**, and nothing else* - otherwise, it's not our problem. Hence why it's also important to attach SHA hashes for the commit you built (or actions workflows binary you pulled from).

As soon as you start making enough changes to necessitate a fork, unless you actually test with *our code and/or binaries specifically*, it's your responsibility.

I'm not trying to be harsh, but it's very clear you're unused to how collaboration works - and it gets especially exhausting when I'm the one being blamed for issues I didn't cause and doesn't affect my work. I literally don't get paid enough to walk people through how FOSS dev works *and* be productive at the same time.