TeamOpenFIRE/OpenFIRE-Firmware #126: [BUG] Rumble Power Limits less than max cause more power-related shutdowns

SeongGino, 2025-06-02:
**Describe the bug**
When rumble is set and enabled, setting the rumble power to anything less than 100% (or `255/255` from the Desktop App) will actually cause *more* power-related shutdowns.

**To Reproduce**
1. Boot up the gun with rumble less than 255, or configure it to use <255 rumble power.
2. Do something to initiate the rumble (either shoot offscreen, use Rumble-based Recoil, or initiate over Serial/an FFB distributor program from the PC)
3. Observe the RP2040 disconnecting and reconnecting erroneously - on some systems, this can appear as `disabling hub (emi?)` or something to that effect.

**Expected behavior**
Rumble activation shouldn't be causing issues regardless, but definitely shouldn't have a higher likelihood when *decreasing* the power/using PWM.

**System Info**
 - Microcontroller board: `rpipico` (but should affect *any*)
 - OpenFIRE Firmware version: `v6.0-`af80a13

**Additional context**
FWIW, if running into this issue even with 100% power, initial tests have shown that adding a 1000uF capacitor in the last step of the rumble driver circuit (where the motor connects to the board) can dramatically improve rumble stability. Rumble stability in general continues to be a challenge with the RP2040 boards - this is just one part that is actually caused in part by software.

roleoroleo, 2025-06-27:
Same problem here, with released 6.0 version.
To reproduce the problem I'm using OpenFIRE App Public Release 3.0.
When I click the test button once the problem doesn't happen.
If I click repeatedly RP2040 crashes.
I tried two different hw configurations: power supply directly from the VBUS and power supply from an external source. Same behavior.
Please, let me know if you need help testing something.

roleoroleo, 2025-10-16:
@aljassim 
Could you describe your setup? Board, power supply, rumble power stage...

roleoroleo, 2025-10-16:
What type of joy motor? Xbox360?

d4v1dk0n7, 2026-01-07:
Hi guys

I'm also experiencing issues with the rumble.

When using the rumble motor as the primary force and disabling the solenoid, the system becomes unstable during intensive trigger usage.

Under rapid or repeated firing, the Raspberry Pi Pico appears to freeze for about 1–2 seconds. During this time:

- The rumble motor stays permanently ON
- The device does not respond to trigger input
- After ~2 seconds, it recovers by itself

The issue does not occur immediately, only after intensive use.

I assume that my issue is related to the same bug you all are experiencing?

roleoroleo, 2026-01-12:
Maybe.
I don't know if the board recovers by itself because I tested the issue only with OpenFIRE App.
In this case, the USB connection dies and I need to close/reopen the serial port.